hallard/WeMos-Matrix-Shield #2: Wiring / Assembling issues

anthonyjclarke, 2018-10-01:
Ordered some 1.2 boards and not having any success wiring up, spent a few hours on this.

Wondering if you have used / have any photos to share?

Seems, I think, my issue is the J1 connector.  The pins on the PCB :- 

[image]

Dont seem to align with Hari's image from his instructable site,

[image]

Is there an issue here as I cannot see R0?

hallard, 2018-10-01:
I do not have received yet the 1.2 boards, so I can't test at all
The J1 connector should go straight to the output of the Matrix Board with a 6 pin ISP Connector 
 
[image]

And yes the trick is that it's not aligned (was my error on PCB 1.0)
As you can see, J1 is matrix in pinout going to matrix out
in the example:

 - J1 pin 1 (so R1 for matrix in) goes to matrix out pin R0 according the following table
 - J1 pin 6 (so B0 for matrix in) goes to matrix out pin G1 according the following table

[image]

I must admit it gave me some headaches ;-)

anthonyjclarke, 2018-10-01:
Thanks. I will try again tomorrow. 

Do you have any photos showing a built PCB and all wired up?

anthonyjclarke, 2018-10-01:
I had to buy 10 PCB's, dont need them all, to thank you for your effort I'm happy to post you some if you want to DM your postal address? Can send today...

dragondaud, 2018-10-05:
Verified: adding wire from G1 input to the pin labeled G1 (which goes to G0) on output fixes board, and it now works great.

hallard, 2018-10-06:
@dragondaud, @anthonyjclarke 
thanks for reporting, I fixed this error. Amazing checked on V1.1 boards, and trace was there, a bad cut on eagle when creating V1.2 and voilà. Here is the visual fix

[image]

Anyway it's now fixed in V1.3, links and documentation has been updated and also PCBs.io boards.

@dragondaud  just saw your firmware, all I wanted to do, so thanks a lot, I'll try it ASAP

Thanks to @anthonyjclarke for the PCB, but I already have a lot here, so I don't need them  ;-)

closed 2018-10-21 by hallard