hallard/WeMos-Matrix-Shield #6: pcbs.io gone?

dragondaud, 2019-09-24:
I placed an order for more of your boards on July 26th with pcbs.io. They took my money and sent email stating "Your PCBs.io Order has been placed! Your order will ship in about 10-14 days. We will send you a shipment email when this occurs."

They have not shipped the order, do not respond to emails, and haven't responded to PayPal dispute messages.

FYI

hallard, 2019-09-24:
Hi

That’s a shame, what’s the status of your order when you connect to your account ? I Placed an order last week

But you see some of my old orders are still in fabrication and I received them, that’s so strange

hallard, 2019-09-24:
I got some V1.4 and V1.5 spare around there, give me your postal address, I will arrange shipment.

dragondaud, 2019-09-27:
This order was placed on Jul 26, 2019, 2:47 PM, and I'v emailed several times with no response.
[image]
I don't have an urgent need, but wanted to try the v1.6 board. I'll just send an order elsewhere. :)

dragondaud, 2019-09-27:
I just checked, and you are right about status not updating. Previous order I did receive does still show In Fabrication, too. They haven't responded to PayPal, either, so PayPal will take my money back from them, sometime soon.

helmarw, 2019-10-26:
i ordered mine here: https://aisler.net
21€ for 3

anthonyjclarke, 2019-10-31:
I went hunting for my 1.6 boards, sure I ordered them... I did!  Same for me, never arrived... did @dragondaud have any success in a refund?

[image]

[image]

dragondaud, 2019-11-01:
I did receive my refund from PayPal, never heard anything from pcbs.io, nor
did PayPal.

hallard, 2020-08-08:
Guys,

just to let you know that PCBs.io are still working, I received a big batch of PCB and the v1.7 is on the road.

closed 2020-08-08 by hallard

Alexisgm7, 2020-09-09:
Hello everyone, I came across this conversation and I am glad to hear that PCBs.io are still working as Hallard said but I tried to make an order today September 9 and their paying platform gives an error as I can't pay with Paypal nor with a credit card.

This is the message that appears: Failed to connect to Paypal. You have not been charged. Please try again.

Does someone have the same issue?  I have already sent them a message, I will have to wait for their response. 

Cheers

loeffma, 2020-09-23:
Hello from Germany.

I have the same problem. Their paying platform gives an error as I can't pay with Paypal nor with a credit card. 
@Alexisgm7 : Do you get any response from PCBs.io?
@hallard : Where can I get the PCB for "LoRa shield for WeMos D1 mini boards" also?

Best regards

Alexisgm7, 2020-09-23:
Hello!

No, I haven't received any reply. And I contacted them via 3 different email adresses I found. 

Cheers

loeffma, 2020-09-23:
Okay. And what does that mean? Is pcbs.io "dead"?
I also don't find an adress on the homepage. I think that's strange ...

hallard, 2020-09-23:
I'm afraid they're down, I looked at my reward and don't see any new order since august 10th, got 3 orders waiting for panelization, so if you're un hurry I suggest moving to oshpark.

loeffma, 2020-09-23:
Do you have a link to the "LoRa shield for WeMos D1 mini boards" on oshpark?

hghile, 2020-09-24:
I've been ordering from pcbs.io on and off over the years, and I've emailed them on multiple occasions and have never ever gotten a single response.  It is incredibly hard to tell if they are still functional.  A few months back, probably a bit earlier than this issue opened, I attempted to order some boards and there was a javascript error preventing payment from loading.  I sent them an email about it (no answer).  I used developer tools and worked around the error, and submitted an order, it appeared to go through.  I sent them an email about how to fix their error.  (no answer) I waited a few weeks, status didn't change, I figured they were dead.  But, a week later, the status said the board went to fab!  then a week later it was back to saying awaiting panelization.  But then it went back to fab, and maybe 5 weeks after placing the order, boards showed up.  And the javascript error got fixed.  pcbs.io was still functional!
But now I can't place orders again.  I hope they aren't gone.  They're good quality and cheap.  oshpark service is way better, but I like saving a few bucks.  I could email them and ask if they're going to fix the payment system, but I expect no answer.

feiticeir0, 2021-02-22:
I'm trying to place an order and I can't... Paypal or credit card is not working... Is there another place to order this PCB ?

hallard, 2021-02-22:
You can order to oshpark, I need to update readme with that link
https://oshpark.com/shared_projects/bzjXuzgQ

If you're in a hurry I got some spare here in France.

Sean-Ocansey, 2021-04-14:
Do you guys know if this service will ever come back? I would really like some Black and gold 1.2mm PCB at a good price. I heard the owner was thinking about selling the business, do you know if there are any buyers interested?

frederic-kleinert, 2022-12-11:
> Ich habe meine hier bestellt: https://aisler.net 21€ für 3 Sie können diesen Link verwenden: https://aisler.net/p/DRGKVMVA

Hallo!
Wie bestellt man bei Aisler? Ich komme nicht zurecht.
Danke!

helmarw, 2022-12-12:
WTF, looks like they broke something. i cant order some of my own projects (which i ordered already a while ago) 
and some others i can.
I do not recommend Aisler anymore, we had a lot of problems with them last year.
Their QM sucks, at least it did a year and a half ago, and it was with Assembly, had really a fun time with that :/
that said simple PCBs are ok more or less, you get what you pay for. 

1)one quick workaround i found is refreshing the webpage a couple of times and suddenly i could click on order.
2)if this doesn't work, after clicking import, download the file first (symbol left upper corner), maybe rename it
and add it as you own project, still might need to ways a bit after upload and refresh webpage.
3)take the gerber.zip form this repository and add it to your Aisler profile, same rules apply, wait a bit and refresh

frederic-kleinert, 2022-12-14:
Danke!
Die Bestellung hat endlich funktioniert.
Wenn nun die Lieferung im Januar auch eintrifft, ist alles in Ordnung.