hallard/WeMos-Matrix-Shield #8: Error WiFi  ...no matching function for call to 'WiFiManager::WiFiManager(Stream&)'

globus1000, 2020-03-14:
Howdy!
I've been trying to compile the myclock project for days. No matter what I try, the compilation always ends with this error in WiFimanager.

.
.
.
_C:\Users\Juergen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

E:\MyClock\Sketch\myClock\WiFi.ino: In function 'void startWiFi()':

WiFi:41:30: error: no matching function for call to 'WiFiManager::WiFiManager(Stream&)'

   WiFiManager wifiManager(OUT);

                              ^

E:\MyClock\Sketch\myClock\WiFi.ino:41:30: note: candidate is:

In file included from E:\MyClock\Sketch\myClock\myClock.ino:13:0:

C:\Users\Juergen\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:78:5: note: WiFiManager::WiFiManager()

     WiFiManager();

     ^

C:\Users\Juergen\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:78:5: note:   candidate expects 0 arguments, 1 provided

WiFi:45:15: error: 'class WiFiManager' has no member named 'setClass'

   wifiManager.setClass("invert");               // dark theme
.
.
.
no matching function for call to 'WiFiManager::WiFiManager(Stream&)'_

I am desperate

I use all files from this site:
https://github.com/dragondaud/myClock
Can you help me?

With kind regards

dragondaud, 2020-04-24:
That error is caused by not using the development version of WiFiManager. The build.sh update script will install the correct version if you use "-u" to update libs. I recently updated it to also use new versions of core and other libs, too.

frederic-kleinert, 2020-04-27:
Thank you, thank you!
 You've helped me a lot. It never occurred to me to look in the development area.
Now the compilation went through and I get a time display. But two things are not so good yet:
1. The clock is two hours slow. 1 hour probably because daylight saving time is not taken into account, but the other hour? Europe/Berlin is given as country.
There is still the possibility to enter a timecode in the web interface, but where do I get it from?
2. in the left corner an error code is displayed in red: 401 and sometimes 404. if I am right, this indicates missing weather data, because the key I entered in OpenWeatherMap is not accepted.

With kind regards

J. Zimmer

Translated with www.DeepL.com/Translator (free version)

frederic-kleinert, 2020-05-01:
Hello!
You can forget my last question, I found everything. The clock was running for a while, but when I wanted to show my grandson how to connect the clock to the wlan via the web-interface, it didn't work anymore. The web-interface does not transfer data anymore. I can tell that because when I call save config in the interface, the "config.json" is empty.  It is also a different interface, because e.g. the color selection looks different.
When I restart apply config, above the clock (it is running, without timecode and without weather data with the wrong zip code and in english) in red letters appears: "config failed". That was it.

With kind regards

Translated with www.DeepL.com/Translator (free version)

frederic-kleinert, 2020-05-01:
[image]

hallard, 2020-08-08:
I think this one is not related to this shield but on the clock software, should be fixed with latest version.

closed 2020-08-08 by hallard