hallard/WeMos-Matrix-Shield #9: Panel is not working - issues with assembly - no instructions ?

feiticeir0, 2021-04-22:
Hi Hallard ! 

First, thank you for the shield. I've ordered 3 from OSHPark. They are version 1.6

I've assembled one, with what I could gather from github - both from version with and without DMA. 
I'm using Wemos D1 mini ESP8266. 
I just see a random streaks of lines on the display. 
I've used 32x32 and 64x64 . I have 32x64, but haven't tested them. 

I've tested with PXMatrix (the ESP32-HUB75-MatrixPanel-I2S-DMA is just for ESP32). I don't have any ESP32 Wemos like the ones you've described . I do have ESP32, but not wemos. 

Any hint what could it be or how to assemble this version 1.6 ? 
I've used - don't know if it's ok or not, 10K resistor and 4.7K resistors. soldered an LDR, a 100uF capacitor (haven't soldered this one yet - it's placed there - because there's no hint on what's the negative side and the positive one - at least none that I can see). 

Thank you

hallard, 2021-04-22:
Would be helpfull to help if you could post a picture of what you've done.

This version works only with PXMatrix because wiring is not for ESP32-HUB75-MatrixPanel-I2S-DMA (even if you use an ESP32). For use [this version](https://github.com/hallard/WeMos-Matrix-Shield-DMA)

To do tests, you just need to solder the headers no components needed (except ESP of course). Better to add caps if ESP goes into reset.

feiticeir0, 2021-04-25:
Hi Hallard. 
I'm going to close the issue. 

I was hopping for some feedback, but I managed to solve this one.  The assembly is well done. My problem is that I had some wemos d1 mini with the sleep solder blob on and that was messing the code and the wemos d1 somehow.  Removing the solder on the PAD solve the issues.  I discover because I've tried with another wemos I have around and worked perfectly. I've tested with 32x32 and 64x32 panels and both worked wonderfully. 
Thank you

closed 2021-04-25 by feiticeir0

hallard, 2021-04-25:
Glad you had it working.

You need to remove big capacitor, it will really slow down light change, this one should be 100nF (unpolarized) only, but you also can remove completely and just let 10K resistor and LDR.